espressif/esp-sr #123: 在调用multinet->destroy 时发生了 assert failed: block_next tlsf_block_functions.h:94 (!block_is_last(block))错误

Z3ce, 2024-11-23:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

可以正常释放

### Actual behavior (suspected bug)

在调用multinet->destroy 时发生了 assert failed: block_next tlsf_block_functions.h:94 (!block_is_last(block))错误

### Error logs or terminal output

```shell
W (29796) m_audio_rec: ## voice read exit
I (29797) m_audio_rec: rec voice destroyed
E (29801) transport_ws: Sec-WebSocket-Accept not found
E (29812) websocket_client: esp_transport_connect() failed with -1, transport_error=ESP_OK, tls_error_code=0, tls_flags=0I (29822) RECORDER_SR: Fetch task destroyed!
, esp_ws_handshake_status_code=502, errno=119
E (29825) ws_audio: WebSocket audio encountered an error, server or network issue!
I (29835) RECORDER_SR: Feed task destroyed!
I (29838) websocket_client: Reconnect after 1000 ms
I (29847) AUDIO_RECORDER: Recorder task destroyed
E (29848) ws_audio: WebSocket audio disconnected, server closed the connection!
I (29849) AUDIO_RECORDER: Recorder destroyed
I (29993) AFE_SR: afe_se_task quit

I (29993) AFE_SR: afe_mase task destroy finished

I (29995) AFE_SR: exit function: afe_destory

assert failed: block_next tlsf_block_functions.h:94 (!block_is_last(block))

Backtrace: 0x403761ae:0x3fcd2cf0 0x403854c1:0x3fcd2d10 0x4038d9bd:0x3fcd2d30 0x4038c2cc:0x3fcd2e60 0x4038c130:0x3fcd2e80 0x40376c0a:0x3fcd2ea0 0x4038d9ed:0x3fcd2ec0 0x42091ff4:0x3fcd2ee0 0x42082821:0x3fcd2f00 0x42082b25:0x3fcd2f30 0x420170f2:0x3fcd2f60 0x42017414:0x3fcd2f80 0x4200cc68:0x3fcd2fb0 0x4200ce05:0x3fcd2fe0 0x4200cfa5:0x3fcd3000 0x4038613d:0x3fcd3090
0x403761ae: panic_abort at D:/.espressif/frameworks/esp-idf-v5.2.1/components/esp_system/panic.c:472
0x403854c1: esp_system_abort at D:/.espressif/frameworks/esp-idf-v5.2.1/components/esp_system/port/esp_system_chip.c:93
0x4038d9bd: __assert_func at D:/.espressif/frameworks/esp-idf-v5.2.1/components/newlib/assert.c:81
0x4038c2cc: block_next at D:/.espressif/frameworks/esp-idf-v5.2.1/components/heap/tlsf/tlsf_block_functions.h:94 (discriminator 1)
 (inlined by) block_link_next at D:/.espressif/frameworks/esp-idf-v5.2.1/components/heap/tlsf/tlsf_block_functions.h:101 (discriminator 1)
 (inlined by) block_mark_as_free at D:/.espressif/frameworks/esp-idf-v5.2.1/components/heap/tlsf/tlsf_block_functions.h:109 (discriminator 1)
 (inlined by) tlsf_free at D:/.espressif/frameworks/esp-idf-v5.2.1/components/heap/tlsf/tlsf.c:1121 (discriminator 1)
0x4038c130: multi_heap_free_impl at D:/.espressif/frameworks/esp-idf-v5.2.1/components/heap/multi_heap.c:234
0x40376c0a: heap_caps_free at D:/.espressif/frameworks/esp-idf-v5.2.1/components/heap/heap_caps.c:393
0x4038d9ed: free at D:/.espressif/frameworks/esp-idf-v5.2.1/components/newlib/heap.c:39
0x42091ff4: model_destroy at /home/sunxiangyu/workspace/esp_sr_lib/components/multinet/multinet7_quantized.c:335
0x42082821: recorder_sr_clear at D:/.espressif/frameworks/esp-adf/components/audio_recorder/recorder_sr.c:492
0x42082b25: recorder_sr_destroy at D:/.espressif/frameworks/esp-adf/components/audio_recorder/recorder_sr.c:578
0x420170f2: audio_rec_stop_recorder at D:/.espressif/frameworks/damengv1.0/project/components/audio_rec/src/audio_rec.c:502
0x42017414: audio_rec_deinit at D:/.espressif/frameworks/damengv1.0/project/components/audio_rec/src/audio_rec.c:621
0x4200cc68: user_switch_work_mode at D:/.espressif/frameworks/damengv1.0/project/main/user_bsp/user_bsp.c:529
0x4200ce05: button_event_callback at D:/.espressif/frameworks/damengv1.0/project/main/user_bsp/user_bsp.c:176
0x4200cfa5: user_ctl_task at D:/.espressif/frameworks/damengv1.0/project/main/user_bsp/user_bsp.c:683
0x4038613d: vPortTaskWrapper at D:/.espressif/frameworks/esp-idf-v5.2.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

### Steps to reproduce the behavior

按照下面上下文运行初始化代码后运行销毁部分代码就出现了此问题

### Project release version

esp-idf V5.2.1发行版本  esp-adfV2.6发行版本 esp-sr TAGV1.7.1

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows10

### Shell

CMD

### Additional context

   附：初始化部分代码
    recorder_sr_cfg_t recorder_sr_cfg = DEFAULT_RECORDER_SR_CFG();
    recorder_sr_cfg.afe_cfg.memory_alloc_mode = AFE_MEMORY_ALLOC_MORE_PSRAM;
    recorder_sr_cfg.afe_cfg.wakenet_init = AUDIO_REC_WAKENET_ENABLE;
    recorder_sr_cfg.multinet_init = AUDIO_REC_MULTINET_ENABLE;
    recorder_sr_cfg.afe_cfg.aec_init = RECORD_HARDWARE_AEC;
    recorder_sr_cfg.afe_cfg.agc_mode = AFE_MN_PEAK_NO_AGC;
    recorder_sr_cfg.afe_cfg.vad_mode = VAD_MODE_2;
    recorder_sr_cfg.feed_task_stack = 4 * 1024;
    recorder_sr_cfg.fetch_task_stack = 4 * 1024;

#if ((CONFIG_ESP32_S3_KORVO2_V3_BOARD == 1) || (CONFIG_ESP32_S3_DAMENG_VOICE_BOARD == 1)) && (CONFIG_AFE_MIC_NUM == 1)
    recorder_sr_cfg.afe_cfg.pcm_config.mic_num = 1;
    recorder_sr_cfg.afe_cfg.pcm_config.ref_num = 1;
    recorder_sr_cfg.afe_cfg.pcm_config.total_ch_num = 2;
    recorder_sr_cfg.input_order[0] = DAT_CH_0;
    recorder_sr_cfg.input_order[1] = DAT_CH_1;

    es7210_mic_select(ES7210_INPUT_MIC1 | ES7210_INPUT_MIC3);
#endif

    audio_rec_cfg_t cfg = AUDIO_RECORDER_DEFAULT_CFG();
    cfg.read = (recorder_data_read_t)&audio_rec_input_cb_for_afe;
    if(s_rec_desc.recorder_sr == NULL)
        s_rec_desc.recorder_sr = recorder_sr_create(&recorder_sr_cfg, &cfg.sr_iface);
    cfg.sr_handle = s_rec_desc.recorder_sr;
#if AUDIO_REC_SPEECH_CMDS_RESET
    if (s_rec_desc.command_word == NULL)
    {
        ESP_LOGW(TAG, "## command word is empty");
    }
    else
    {
        char err[400];
        recorder_sr_reset_speech_cmd(cfg.sr_handle, s_rec_desc.command_word, err);
    }
#endif

    cfg.event_cb = audio_rec_engine_event_cb;
    cfg.wakeup_time = AUDIO_REC_WAKEUP_TIMEOUT;
    cfg.vad_start = AUDIO_REC_VAD_SPEAK_TIME;
    cfg.vad_off = AUDIO_REC_VAD_SILENCE_TIME;
    cfg.wakeup_end = AUDIO_REC_WAKEEND_TIMEOUT;
    // cfg.wakeup_end = AUDIO_REC_WAKEUP_TIMEOUT;
    s_rec_desc.recorder = audio_recorder_create(&cfg);
    return s_rec_desc.recorder;
    
    销毁部分代码
    if (s_rec_desc.recorder)
    {
        ret = audio_recorder_destroy(s_rec_desc.recorder);
        if (ret != ESP_OK)
        {
            ESP_LOGE(TAG, "Failed to destroy audio recorder: %s", esp_err_to_name(ret));
        }
        s_rec_desc.recorder = NULL;
    }

    if (s_rec_desc.recorder_sr)
    {
        ret = recorder_sr_destroy(s_rec_desc.recorder_sr);
        if (ret != ESP_OK)
        {
            ESP_LOGE(TAG, "Failed to destroy recorder_sr: %s", esp_err_to_name(ret));
        }
        s_rec_desc.recorder_sr = NULL;
    }
    相关配置
    
[sdkconfig.txt](https://github.com/user-attachments/files/17878355/sdkconfig.txt)
此现象只会在 multinet选择mn7_cn和mn6_cn上出现在mn5q8_cn不会出现

sun-xiangyu, 2024-11-25:
@Z3ce , please update esp-sr to the latest version and try again.

Z3ce, 2024-11-25:
@sun-xiangyu Ok I tested V1.9.4 and the master branch and this problem still exists
[image]

sun-xiangyu, 2024-11-26:
I can not reproduce your issue. I tested the destroy function for mn7_cn using [this test code](https://github.com/espressif/esp-sr/blob/master/test_apps/esp-sr/main/test_multinet.cpp#L25-L65), and it works fine. Could you please try to reproduce your problem based on this code?

Z3ce, 2024-11-29:
OK  I think it's a question of where do I call it and let me test it this way

ho-229, 2024-12-01:
Hi @sun-xiangyu , I encounter the same problem. I think the [test code](https://github.com/espressif/esp-sr/blob/master/test_apps/esp-sr/main/test_multinet.cpp#L25-L65) is incomplete, maybe we should modify command words or feed audio data in our test.

sun-xiangyu, 2024-12-03:
Hi @ho-229 , there are many test cases in [test_multinet.cpp](https://github.com/espressif/esp-sr/blob/master/test_apps/esp-sr/main/test_multinet.cpp). The situation you mentioned has already been tested by us. It is best to attach a your case.

PR #150 (2025-05-13): bugfix: fixed #148 (AIS-1999)

## Description

## Related

## Tes
closed 2025-06-11 by Z3ce