espressif/esp-sr #126: AFE Memory footprint improvement

boborjan2, 2025-01-17:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

We are using AFE (AEC/AGC/NS) in a bluetooth classic application (no wifi/lwip). Target is ESP32 with 2M spiram. AFE version: VC_V220727. (esp-sr is of date 2024.09.04)
All key functions (AEC,AGC,NS) are functional, we could set them up, they seem to work OK.
However internal RAM consumption is very critical, we are unable to enable the bluetooth profiles we need because of lack of memory. I turned on panic/abort on memory allocation failure and the issue is quite clear. When connecting on bluetooth, system panics on a 32byte malloc in internal memory. As an info,  CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=4096
**AFE uses 121kB of internal mem (+ approx 140k spiram).** 
We followed the guidelines for saving memory. I checked all seemingly relevant config option is sdkconfig.
We are not using ADF at all but AFE directly.

Generally it would be great if AFE used less internal memory. 

We beleive there would be room for adjusting afe ram footprint as well, e.g. 
- fetch method could have a max time to wait parameter. This way there would not be the need for another task (with stack in internal ram) just to read data. This task needs a ringbuffer just pass data to it's sink (the ringbuffer is already in spiram). Btw, for how long is data in fetch struct valid?
This takes only a few k, but even with a 2M ESP32, every k of internal memory matters.

### Use cases

Generally it would be great if AFE used less internal memory.

boborjan2, 2025-01-17:
Our AFE config is:
`
       afe_config_t afe_config = AFE_CONFIG_DEFAULT();
        afe_config.vad_init = false;
        afe_config.wakenet_init = false;
        afe_config.afe_mode = SR_MODE_HIGH_PERF;
        afe_config.afe_perferred_core = 0;
        afe_config.afe_perferred_priority = 21; //TBD
        afe_config.memory_alloc_mode = AFE_MEMORY_ALLOC_MORE_PSRAM;
        afe_config.pcm_config.mic_num = 1;
        afe_config.pcm_config.ref_num = 1;
        afe_config.pcm_config.total_ch_num = 1 + 1;
        afe_config.afe_ns_mode = NS_MODE_SSP;

        afe_config.pcm_config.sample_rate = HFP_SAMPLE_RATE;
        afe_config.voice_communication_init = true;

        bt_i2s.afe_handle = &ESP_AFE_VC_HANDLE;

        afe_config.vad_init = false;
        afe_config.aec_init = true;
        afe_config.se_init = true;
        afe_config.voice_communication_agc_init = true;
        afe_config.voice_communication_agc_gain = 15;
`

feizi, 2025-01-20:
@boborjan2 , thanks for your feedback, we will update AFE pipeline to make it more efficient.