espressif/esp-sr #127: 成功运行了唤醒词检测但没有data_volume输出

PawnMa, 2025-03-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

代码如下，唤醒词唤醒正常，但data_volume的打印一致是0，请问可能是什么原原因
```
void AudioAFE::AudioFrontEndTask() {
    int fetch_size = afe_handle_->get_fetch_chunksize(afe_data_);
    int feed_size = afe_handle_->get_feed_chunksize(afe_data_);
    ESP_LOGI(TAG,
             "Audio communication task started, feed size: %d fetch size: %d",
             feed_size, fetch_size);
    while (true) {
        afe_fetch_result_t* res = afe_handle_->fetch(afe_data_);
        if (res->wakeup_state == WAKENET_DETECTED) {
            ESP_LOGI(TAG, "model index:%d, word index:%d\n",
                     res->wakenet_model_index, res->wake_word_index);
        }
        if (res == nullptr || res->ret_value == ESP_FAIL) {
            if (res != nullptr) {
                ESP_LOGI(TAG, "Error code: %d", res->ret_value);
            }
            continue;
        }
        ESP_LOGI(TAG, "data_volume: %d", res->data_volume);
    }
}
```

PawnMa, 2025-03-04:
补充初始化设置
```
void AudioAFE::Initialize(int channels, bool reference) {
    channels_ = channels;
    reference_ = reference;

    srmodel_list_t* models = esp_srmodel_init("model");
    afe_config_t* afe_config =
        afe_config_init("MR", models, AFE_TYPE_SR, AFE_MODE_LOW_COST);
    afe_config->vad_mode = VAD_MODE_3;

    afe_config = afe_config_check(afe_config);

    afe_config_print(afe_config);
    // 创建api操作句柄
    afe_handle_ = esp_afe_handle_from_config(afe_config);
    // 创建数据结构句柄
    afe_data_ = afe_handle_->create_from_config(afe_config);
    afe_config_free(afe_config);

    xTaskCreate(
        [](void* arg) {
            auto this_ = (AudioAFE*)arg;
            this_->AudioFrontEndTask();
            vTaskDelete(NULL);
        },
        "audio_communication", 4096 * 2, this, 1, NULL);
}
```

feizi, 2025-03-04:
It is a bug. We update the interface of AFE, but do not update the function to calculate `data_volume`.

PawnMa, 2025-03-04:
非常感谢您的回答，期待能尽快修复

closed 2025-03-04 by PawnMa

stintel, 2025-04-24:
Can this be reopened until it is actually fixed?