espressif/esp-sr #128: flash的内存放不下deep noise suppressionV2了

Freak-luo, 2025-03-06:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

ESP Speech Recongnition 的选项看到可以Read model data from SD Card   我把Deep noise Suppression V2的模型放进SD卡里后，我的SD卡模型要如何命名  存放的路径要如何填写？

[image]

sun-xiangyu, 2025-03-11:
Hi @Freak-luo , 现在` Read model data from SD Card ` 主要是我们使用作为 host test的，目前在芯片上还用不了

Xinyu-Sheng, 2025-06-29:
> Hi [@Freak-luo](https://github.com/Freak-luo) , 现在`Read model data from SD Card` 主要是我们使用作为 host test的，目前在芯片上还用不了

有计划将来在芯片上实现吗？