espressif/esp-sr #129: Esp-idf Flash failing with error in esp-sr makefile

gwoonryan, 2025-03-09:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I would expect the "flash" target to work.

### Actual behavior (suspected bug)

The "app" / build target completes successfully, but when I try to flash the board, the process fails with an error.
In the logs, I have added the full "flash" logs and the main error.

### Error logs or terminal output

```shell
Error:
CMake Error at managed_components/espressif__esp-sr/CMakeLists.txt:113 (add_dependencies):
  Cannot add target-level dependencies to non-existent target "flash".

  The add_dependencies works for top-level logical targets created by the
  add_executable, add_library, or add_custom_target commands.  If you want to
  add file-level dependencies see the DEPENDS option of the add_custom_target
  and add_custom_command commands.

Full log:

====================[ Build | flash | Debug ]===================================
C:\Users\ryanv\AppData\Local\Programs\CLion\bin\cmake\win\x64\bin\cmake.exe --build C:\Users\ryanv\Desktop\raket\code\raket-zaken-git-repo\esp_idf_test\cmake-build-debug --target flash -j 6
[1/7] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Users\ryanv\Desktop\raket\code\raket-zaken-git-repo\esp_idf_test\cmake-build-debug\esp-idf\esptool_py && python C:/Espressif/components/partition_table/check_sizes.py --offset 0x8000 partition --type app C:/Users/ryanv/Desktop/raket/code/raket-zaken-git-repo/esp_idf_test/cmake-build-debug/partition_table/partition-table.bin C:/Users/ryanv/Desktop/raket/code/raket-zaken-git-repo/esp_idf_test/cmake-build-debug/origin_flight_computer.bin"
origin_flight_computer.bin binary size 0x68b70 bytes. Smallest app partition is 0x100000 bytes. 0x97490 bytes (59%) free.
[2/7] Move and Pack models...
Recommended model partition size: 3902K
[3/7] Performing configure step for 'bootloader'
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-configure C:/Users/ryanv/Desktop/raket/code/raket-zaken-git-repo/esp_idf_test/cmake-build-debug/bootloader-prefix/src/bootloader-stamp/bootloader-configure 
C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Users\ryanv\Desktop\raket\code\raket-zaken-git-repo\esp_idf_test\cmake-build-debug\bootloader && C:\Users\ryanv\AppData\Local\Programs\CLion\bin\cmake\win\x64\bin\cmake.exe -DSDKCONFIG=C:/Users/ryanv/Desktop/raket/code/raket-zaken-git-repo/esp_idf_test/sdkconfig -DIDF_PATH=C:/Espressif -DIDF_TARGET=esp32s3 -DPYTHON_DEPS_CHECKED=1 -DPYTHON=python -DEXTRA_COMPONENT_DIRS=C:/Espressif/components/bootloader -DPROJECT_SOURCE_DIR=C:/Users/ryanv/Desktop/raket/code/raket-zaken-git-repo/esp_idf_test -DIGNORE_EXTRA_COMPONENT= -GNinja -S C:/Espressif/components/bootloader/subproject -B C:/Users/ryanv/Desktop/raket/code/raket-zaken-git-repo/esp_idf_test/cmake-build-debug/bootloader && C:\Users\ryanv\AppData\Local\Programs\CLion\bin\cmake\win\x64\bin\cmake.exe -E touch C:/Users/ryanv/Desktop/raket/code/raket-zaken-git-repo/esp_idf_test/cmake-build-debug/bootloader-prefix/src/bootloader-stamp/bootloader-configure"
-- Minimal build - OFF
-- Building ESP-IDF components for target esp32s3
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_wifi_remote
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_wifi_remote
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_wifi_remote
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_wifi_remote
NOTICE: Processing 26 dependencies:
NOTICE: [1/26] chmorgan/esp-libhelix-mp3 (1.0.3)
NOTICE: [2/26] espressif/cbor (0.6.0~1)
NOTICE: [3/26] espressif/esp-dsp (1.4.12)
NOTICE: [4/26] espressif/esp-modbus (1.0.17)
NOTICE: [5/26] espressif/esp-serial-flasher (0.0.11)
NOTICE: [6/26] espressif/esp-sr (1.9.5)
NOTICE: [7/26] espressif/esp-zboss-lib (1.6.3)
NOTICE: [8/26] espressif/esp-zigbee-lib (1.6.3)
NOTICE: [9/26] espressif/esp_diag_data_store (1.0.2)
NOTICE: [10/26] espressif/esp_diagnostics (1.2.1)
NOTICE: [11/26] espressif/esp_insights (1.2.2)
NOTICE: [12/26] espressif/esp_modem (1.3.0)
NOTICE: [13/26] espressif/esp_rainmaker (1.5.2)
NOTICE: [14/26] espressif/esp_rcp_update (1.2.0)
NOTICE: [15/26] espressif/esp_schedule (1.2.0)
NOTICE: [16/26] espressif/esp_secure_cert_mgr (2.5.0)
NOTICE: [17/26] espressif/jsmn (1.1.0)
[... 83 lines cut ...]
Flash a project that has esp-sr in the managed components with the target set to esp32s3

### Project release version

1.9.5 (downloaded by component manager)

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

CMD

### Additional context

I don't really understand why esp-idf it is trying to do anything with the esp-sr library, because I am not using it in my code.

sun-xiangyu, 2025-03-11:
By default, esp-idf does not compile esp-sr unless you call esp-sr in idf_component or a component you are using utilizes esp-sr. You can search for the esp-sr directory in your project and then comment out the CMakeLists.txt file under the esp-sr directory. 
From your log, it seems that you are not using a partition table. We will add relevant filtering in CMake in the future.

gwoonryan, 2025-03-11:
I was indeed using the esp32-arduino component, that is why it was compiling esp-sr