espressif/esp-sr #134: Request for branch or version including esp_afe_vc_iface.h (VC feature)

interRyoko, 2025-04-01:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi Espressif team,

I am currently working with ESP-IDF v5.4 and would like to use the Voice Communication (VC) feature of the esp-sr component.

I noticed that some references mention the file `esp_afe/include/esp_afe_vc_iface.h`, but I couldn’t find this file in any of the following branches:

- `master`
- `dev/p4`
- `feat/add_afe_volume`
- `feat/support_two_words`
- `tts_support_3.3`
- ...and others

Could you kindly clarify which branch or upcoming version of `esp-sr` includes VC support and this header file?

If it's not publicly available yet, is there a recommended way to obtain or test this functionality?

Thank you very much for your support!

Best regards,
Ryoko

sun-xiangyu, 2025-04-01:
Hi @interRyoko 

You can find  `esp_afe_vc_iface.h` in esp-sr [v1.9.5](https://github.com/espressif/esp-sr/tree/v1.9.5).
Now we remove this header file, but you can use afe_vc by the new afe interface, like this 
```
srmodel_list_t *models = esp_srmodel_init("model");
afe_config_t *afe_config = afe_config_init("MMNR", models, AFE_TYPE_VC, AFE_MODE_HIGH_PERF);
```

Please refer to [document](https://docs.espressif.com/projects/esp-sr/en/latest/esp32s3/audio_front_end/README.html) and [example](https://github.com/espressif/esp-skainet/tree/master/examples/voice_communication) for more details.

interRyoko, 2025-04-02:
Hi @sun-xiangyu 
Thank you for the clarification!

We'll switch to using the new AFE interface with `AFE_TYPE_VC` as you suggested.  
Just to confirm: using `afe_config_init("MMNR", models, AFE_TYPE_VC, AFE_MODE_HIGH_PERF)`  
will properly initialize VC functionality even without `esp_afe_vc_iface.h`, right?

Also, could you point us to the specific example or documentation section that demonstrates this usage?

Thanks again for your support!

closed 2025-04-02 by interRyoko