espressif/esp-sr #135: IDFv5.4使用SRv2.0.3的afe_config_init出现找不到模型(已启用hi lexin)

zggcd157, 2025-04-02:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

我的预期为初始化后哦打印模型name

### Actual behavior (suspected bug)

我抄官方的示例、

[image]

教程和sdkconfig都无法解决这个错误（）

后来我更改了分区表为factory(10M)， ota_0/_1(1M)， model(6M)警告才消失，之前ota_0/_1分别为6M，我的flash没有溢出(我是32Mflash),

一旦我的ota_0/_1大于3M就会重新触发这个警告(把ota_0/_16M，factory改为4M也不行，ota_0/_1不必须3M以下)

### Error logs or terminal output

```shell
W： AFE_CONFIG: wakenet model not found. please load wakenet model...
```

### Steps to reproduce the behavior

afe_config_init

### Project release version

v2.0.3

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

win11

### Shell

other (details in Additional context)

zggcd157, 2025-04-02:
我打错了几个字，下面是修正

把ota_0/_1 6M，factory改为4M也不行，ota_0/_1必须3M以下）

sun-xiangyu, 2025-04-02:
一个唤醒词模型的大小在300KB 以内，如果你只需要加载唤醒词模型，可以尝试将model分区设置为1MB

zggcd157, 2025-04-06:
我在不开启CONFIG_SR_MN_CN_MULTINET7_QUANT的情况下会不会影响我后续的SR的使用？，我根据让readme和上网找教程都没有发现对应CONFIG_SR_MN_CN_MULTINET7_QUANT的解释

我在不开CONFIG_SR_MN_CN_MULTINET7_QUAN的情况下可以正常唤醒和识别(目前只进行的HI 乐鑫、小爱同学的唤醒测试)

sun-xiangyu, 2025-04-07:
`CONFIG_SR_MN_CN_MULTINET7_QUANT` 是命令词模型，就是打开空调，关闭空调这些命令词。不会影响唤醒和识别。