espressif/esp-sr #137: esp32p4 + esp-idf5.4.1 build error

logicDance-mcu, 2025-04-03:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

正确编译

### Actual behavior (suspected bug)

[image]
这个编译不通过

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

开启 sr 语音识别

### Project release version

ESP-IDF 5.4.1   ESP-ADF 2.7

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

ubuntu 24.04

### Shell

Bash

sun-xiangyu, 2025-04-07:
I will try this esp-idf.

sun-xiangyu, 2025-04-07:
I'm using the latest esp-sr, and I haven't encountered the issue you mentioned. What version of esp-sr are you using?

closed 2026-04-03 by feizi