espressif/esp-sr #139: IDF5.4.1 使用SRv2.0.3一旦启用CHINESE_SR_MN_MODEL_SEL的模型会导致看门狗复位

zggcd157, 2025-04-06:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

预期和官方示例一致全绿日志和退出app_main

### Actual behavior (suspected bug)

我抄的esp-skainet-master\examples\cn_speech_commands_recognition提供的初始化流程，使用的partitions.cvs
如图片所示

在执行到afe_handle->create_from_config(afe_config);导致程序复位，我也尝试过deinit看门狗(2分钟以上)程序还是在afe_handle->create_from_config卡住不退出，由于ESP-SR没有提供afe_handle->create_from_config的实现.c我无法查看阻塞原因

[image]

[image]

[image]

[image]

[image]

### Error logs or terminal output

```shell
I (839) MODEL_LOADER: The storage free size is 25088 KB
I (849) MODEL_LOADER: The partition size is 6144 KB
I (849) MODEL_LOADER: Successfully load srmodels
I (859) AFE_CONFIG: Set VAD Model: vadnet1_medium
I (859) AFE_CONFIG: Set WakeNet Model: wn9_hilexin
I (869) AFE_CONFIG: Set Second WakeNet Model: wn9_hilexin
进入afe_handle->create_from_config
E (5839) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (5839) task_wdt:  - IDLE0 (CPU 0)
E (5839) task_wdt: Tasks currently running:
E (5839) task_wdt: CPU 0: main
E (5839) task_wdt: CPU 1: IDLE1
E (5839) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x42054917:0x3FC9D7C0 0x4037C796:0x3FC9D7F0 0x4204FA9F:0x3FCA3E10 0x4201F68F:0x3FCA3E40 0x4200C6E3:0x3FCA4020 0x4200C8D4:0x3FCA4040 0x420097AD:0x3FCA4060 0x42009769:0x3FCA4080 0x42076840:0x3FCA40B0
--- 0x42054917: task_wdt_timeout_handling at E:/Espressif/IDF/v5.4/esp-idf/components/esp_system/task_wdt/task_wdt.c:434
---  (inlined by) task_wdt_isr at E:/Espressif/IDF/v5.4/esp-idf/components/esp_system/task_wdt/task_wdt.c:507
--- 0x4037c796: _xt_lowint1 at E:/Espressif/IDF/v5.4/esp-idf/components/xtensa/xtensa_vectors.S:1240
--- 0x4204fa9f: flash_model_info at /home/sunxiangyu/workspace/esp_sr_lib/components/hufzip/hufzip_model.c:488 (discriminator 1)
--- 0x4201f68f: model_create at /home/sunxiangyu/workspace/esp_sr_lib/components/vadnet/vadnet1_quantized.c:203
--- 0x4200c6e3: afe_init_vad at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr_1mic.c:275
--- 0x4200c8d4: afe_create_from_config at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr_1mic.c:471
--- 0x420097ad: SR_init at E:/Espressif/VSProjects/watches/apps/SR/SR.c:39
--- 0x42009769: app_main at E:/Espressif/VSProjects/watches/main/main.c:29
--- 0x42076840: main_task at E:/Espressif/IDF/v5.4/esp-idf/components/freertos/app_startup.c:208
```

### Steps to reproduce the behavior

在SDKconfig或者menuconfig选择CONFIG_SR_MN_CN_MULTINET7_QUANT或者其他的CONFIG_SR_MN_CN_ 就会在调用afe_handle->create_from_config时触发WDT复位

### Project release version

IDFv5.4.1 SRv2.0.3

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

win11

### Shell

PowerShell

zggcd157, 2025-04-06:
我只使用了Hi 乐鑫和小爱同学

[image]

[image]

zggcd157, 2025-04-06:
下面是我的不选择CHINESE_SR_MN_MODEL_SEL里面的任何选项打印的日志

[image]

[image]

[image]

sun-xiangyu, 2025-04-09:
这应该是esp-idf 的问题，你可以看 [model_path.c ](https://github.com/espressif/esp-sr/blob/master/src/model_path.c#L310)这个就是 esp-sr 加载模型的代码