espressif/esp-sr #142: SR_WN_WN9_XIAOYUTONGXUE_TTS2 wakenet model not found. please load wakenet model...

Rvn0xsy, 2025-04-20:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

我在使用`小宇同学 (wn9_xiaoyutongxue_tts2)`这个模型烧录至固件中启动后，发现提示：
```
AFE_CONFIG: wakenet model not found. please load wakenet model...
```

而我使用其他模型就没有任何问题，例如：`小明同学 (wn9_xiaomingtongxue_tts2`

### Actual behavior (suspected bug)

使用了 `小宇同学 (wn9_xiaoyutongxue_tts2)` 模型的情况：

```
I (4539) MODEL_LOADER: The partition size is 960 KB
I (4539) MODEL_LOADER: Successfully load srmodels
I (4599) AFE: AFE Version: (1MIC_V250121)
I (4599) AFE: Input PCM Config: total 1 channels(1 microphone, 0 playback), sample rate:16000
I (4599) AFE: AFE Pipeline: [input] -> |NS(nsnet2)| -> |VAD(WebRTC)| -> [output]
I (4609) AudioProcessor: Audio communication task started, feed size: 512 fetch size: 512
I (4619) WakeWordDetect: Model 0: nsnet2
I (4619) WakeWordDetect: Model 1: vadnet1_medium
I (4629) AFE_CONFIG: Set Noise Suppression Model: nsnet2
I (4629) AFE_CONFIG: Set VAD Model: vadnet1_medium
**W (4639) AFE_CONFIG: wakenet model not found. please load wakenet model...**
MC Quantized vadnet1:vadnet1_mediumv1_Speech_1_0.5_0.1, min speech:128 ms, min noise:992 ms, mode:0, threshold:0.500, channel:1, tigger:v1 (Apr 17 2025 15:53:26)
I (4679) AFE: AFE Version: (1MIC_V250121)
I (4689) AFE: Input PCM Config: total 1 channels(1 microphone, 0 playback), sample rate:16000
I (4699) AFE: AFE Pipeline: [input] -> |VAD(vadnet1_medium)| -> [output]
I (4709) WakeWordDetect: Audio detection task started, feed size: 512 fetch size: 512
I (4709) Application: STATE: idle
```

使用了 `小明同学 (wn9_xiaomingtongxue_tts2`的情况：

```
I (4489) MODEL_LOADER: The storage free size is 21888 KB
I (4499) MODEL_LOADER: The partition size is 960 KB
I (4499) MODEL_LOADER: Successfully load srmodels
I (4549) AFE: AFE Version: (1MIC_V250121)
I (4549) AFE: Input PCM Config: total 1 channels(1 microphone, 0 playback), sample rate:16000
I (4559) AFE: AFE Pipeline: [input] -> |NS(nsnet2)| -> |VAD(WebRTC)| -> [output]
I (4559) AudioProcessor: Audio communication task started, feed size: 512 fetch size: 512
I (4569) WakeWordDetect: Model 0: nsnet2
I (4579) WakeWordDetect: Model 1: vadnet1_medium
I (4579) WakeWordDetect: Model 2: wn9_xiaomingtongxue_tts2
I (4589) AFE_CONFIG: Set Noise Suppression Model: nsnet2
I (4589) AFE_CONFIG: Set VAD Model: vadnet1_medium
**I (4599) AFE_CONFIG: Set WakeNet Model: wn9_xiaomingtongxue_tts2
MC Quantized vadnet1:vadnet1_mediumv1_Speech_1_0.5_0.1, min speech:128 ms, min noise:992 ms, mode:0, threshold:0.500, channel:1, tigger:v1 (Apr 17 2025 15:53:26)
MC Quantized wakenet9: wakenet9l_tts2h12_小明同学_3_0.626_0.632, tigger:v3, mode:0, p:0, (Apr 17 2025 15:53:24)**
I (4669) AFE: AFE Version: (1MIC_V250121)
I (4679) AFE: Input PCM Config: total 1 channels(1 microphone, 0 playback), sample rate:16000
I (4689) AFE: AFE Pipeline: [input] -> |VAD(vadnet1_medium)| -> |WakeNet(wn9_xiaomingtongxue_tts2,)| -> 2,)| -> [output]
I (4699) WakeWordDetect: Audio detection task started, feed size: 512 fetch size: 512
```

### Error logs or terminal output

```shell
I (4629) AFE_CONFIG: Set VAD Model: vadnet1_medium
**W (4639) AFE_CONFIG: wakenet model not found. please load wakenet model...**
MC Quantized vadnet1:vadnet1_mediumv1_Speech_1_0.5_0.1, min speech:128 ms, min noise:992 ms, mode:0, threshold:0.500, channel:1, tigger:v1 (Apr 17 2025 15:53:26)
I (4679) AFE: AFE Version: (1MIC_V250121)
```

### Steps to reproduce the behavior

```
I (4539) MODEL_LOADER: The partition size is 960 KB
I (4539) MODEL_LOADER: Successfully load srmodels
I (4599) AFE: AFE Version: (1MIC_V250121)
I (4599) AFE: Input PCM Config: total 1 channels(1 microphone, 0 playback), sample rate:16000
I (4599) AFE: AFE Pipeline: [input] -> |NS(nsnet2)| -> |VAD(WebRTC)| -> [output]
I (4609) AudioProcessor: Audio communication task started, feed size: 512 fetch size: 512
I (4619) WakeWordDetect: Model 0: nsnet2
I (4619) WakeWordDetect: Model 1: vadnet1_medium
I (4629) AFE_CONFIG: Set Noise Suppression Model: nsnet2
I (4629) AFE_CONFIG: Set VAD Model: vadnet1_medium
**W (4639) AFE_CONFIG: wakenet model not found. please load wakenet model...**
MC Quantized vadnet1:vadnet1_mediumv1_Speech_1_0.5_0.1, min speech:128 ms, min noise:992 ms, mode:0, threshold:0.500, channel:1, tigger:v1 (Apr 17 2025 15:53:26
```

### Project release version

v2.0.5

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Arch Linux

### Shell

ZSH

### Additional context

我尝试了两个模型，使用了同样的配置，但是唯独 `小宇同学 (wn9_xiaoyutongxue_tts2)`这个模型出现无法找到的错误。

sun-xiangyu, 2025-04-21:
我知道问题了，我会修复这个问题

sun-xiangyu, 2025-04-22:
Please try again.

closed 2025-04-23 by feizi