espressif/esp-sr #143: VAD usage question

theweipeng, 2025-04-22:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

When I knock the table lightly, the esp-sr correctly detects it as `SILENT`. However, if I knock the table loudly and continuously, the esp-sr detects the noise as `SPEECH`. Is this behavior expected? Or is there an issue with my configuration?  My situation is to detect whether the human voice is end , when the voice end I will send the voice to the remote server. So I need to make the detecting  as correct as possible

This is my init code:

```c++
    channels_ = channels;
    reference_ = reference;
    int ref_num = reference_ ? 1 : 0;

    std::string input_format;
    for (int i = 0; i < channels_ - ref_num; i++) {
        input_format.push_back('M');
    }
    for (int i = 0; i < ref_num; i++) {
        input_format.push_back('R');
    }

    afe_config_t* afe_config = afe_config_init(input_format.c_str(), NULL, AFE_TYPE_VC, AFE_MODE_HIGH_PERF);
    afe_config->aec_init = true;
    afe_config->aec_mode = AEC_MODE_VOIP_HIGH_PERF;
    afe_config->vad_init = true;         
    afe_config->vad_min_noise_ms = 500; 
    afe_config->vad_min_speech_ms = 500; 
    afe_config->vad_delay_ms = 128;     
    afe_config->vad_mode = VAD_MODE_0;  
    
    afe_config->afe_perferred_core = 1;
    afe_config->afe_perferred_priority = 1;
    afe_config->agc_init = true;
    afe_config->agc_mode = AFE_AGC_MODE_WAKENET;
    afe_config->agc_compression_gain_db = 10;
    afe_config->memory_alloc_mode = AFE_MEMORY_ALLOC_MORE_PSRAM;

    afe_iface_ = esp_afe_handle_from_config(afe_config);
    afe_data_ = afe_iface_->create_from_config(afe_config);
    
```

sun-xiangyu, 2025-04-22:
If you need speech recognition, it is recommended that you choose AFE_TYPE_SR.  
 If you need a higher-performance VAD, you need to select vadnet through menuconfig. You can refer to [this example](https://github.com/espressif/esp-skainet/blob/master/examples/voice_activity_detection/main/main.c) about how to use vadnet.

theweipeng, 2025-04-22:
> If you need speech recognition, it is recommended that you choose AFE_TYPE_SR. If you need a higher-performance VAD, you need to select vadnet through menuconfig. You can refer to [this [...]

Yes, I followed this doc (https://docs.espressif.com/projects/esp-sr/en/latest/esp32/vadnet/README.html) to select the `vadnet1 medium`. My question is whether the hebavior is  correctly, when I knocking the table loudly and continuously, the esp-sr detects the noise as `SPEECH`.  
Thanks, I will reproduct the example to test this case.

sun-xiangyu, 2025-04-22:
>  afe_config_t* afe_config = afe_config_init(input_format.c_str(), NULL, AFE_TYPE_VC, AFE_MODE_HIGH_PERF);

According to your code, you haven't correctly loaded vadnet. 

The answer to your question is, of course, NO. If it returns SPEECH, that means the model's performance is not good enough, resulting in incorrect recognition.

theweipeng, 2025-04-24:
> According to your code, you haven't correctly loaded vadnet.
>
> The answer to your question is, of course, NO. If it returns SPEECH, that means the model's performance is not good enough, resulting in [...]

Thank you! It worked for me.

closed 2025-04-24 by theweipeng