espressif/esp-sr #144: 使用VScode生成menuconfig时，同名的menu，因为第一个depends不满足，两个都被隐藏。使用shell脚本或cmd方式则不会有以上问题。

nukerl, 2025-04-25:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

OS
Windows

Operating System version
Win11

Visual Studio Code version
1.99.3

ESP-IDF version
5.4.0

Python version
3.8.7

Doctor command output
无

Extension
无

Description
两个同名kconfig文件中的 esp-sr 2.1.0版本中的 menu "Load Multiple Wake Words" 因为第一个depends不满足，两个都被隐藏。
In the two kconfig files with the same name, the "Load Multiple Wake Words" menu in the esp-sr 2.1.0 version is hidden in both cases because the first depends condition is not met.

### Actual behavior (suspected bug)

In the two kconfig files with the same name, the "Load Multiple Wake Words" menu in the esp-sr 2.1.0 version is hidden in both cases because the first depends condition is not met.

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Debug Message
当我尝试调换kconfig文件中的 比如esp-sr 2.1.0版本中的 menu "Load Multiple Wake Words"的顺序，使得第一个menu的depends满足要求时，两个Load Multiple Wake Words都会显现。当我尝试把两个Load Multiple Wake Words更名为Load Multiple Wake Words1和Load Multiple Wake Words2时，Load Multiple Wake Words2也会正确出现。可见，出现问题的原因就是因为两个menu重名了 但是又只判断了第一个depends。建议修改一下。经过测试，在shell脚本或者cmd脚本中生成menuconfig时不会有以上问题。

Other Steps to Reproduce
编译任何带有esp-sr组件的工程均会发现这个问题。近期sr组件更新了2.1.0版本，出现了两个同名的menu "Load Multiple Wake Words"

### Project release version

2.1.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

win11

### Shell

other (details in Additional context)

### Additional context

翻译：
In two kconfig files with the same name, for example, in the esp-sr version 2.1.0, the menu "Load Multiple Wake Words" is hidden for both because the first depends condition is not met.
Debugging message
When I tried to swap the order of the menu "Load Multiple Wake Words" in the kconfig file, for example, in the esp-sr 2.1.0 version, to make the depends of the first menu meet the requirements, both "Load Multiple Wake Words" menus would appear. When I attempted to rename the two "Load Multiple Wake Words" menus to "Load Multiple Wake Words1" and "Load Multiple Wake Words2", "Load Multiple Wake Words2" would also appear correctly. It is evident that the issue arises because the two menus have the same name, but only the depends of the first one is checked. It is suggested to make a modification. After testing, there is no such problem when generating menuconfig in a shell script or a cmd script.
Other reproduction steps
Compiling any project with the esp-sr component will reveal this issue. Recently, the sr component was updated to version 2.1.0, and two menus with the same name, "Load Multiple Wake Words", appeared.

以上问题只在vscode中出现，不知道两个同名的menu是否是合法的命名。因为在shell中正常，我尝试把问题提交给vscode的IDF拓展的github，但他们认为不是他们的问题，应该来找你们。以下是他的回复。
The above issue only occurs in VSCode. I'm not sure if having two menus with the same name is a valid naming. Since it works fine in the shell, I tried to submit the problem to the VSCode IDF extension's GitHub, but they think it's not their issue and suggested I come to you. Here is their reply.

[image]

[image]

Hi @nukerl I don't think this repository is the right place to fix Kconfig issue for you. The reason is that Kconfig resolution is done by which uses the Kconfiglib used in the ESP-IDF build system and the extension just rely that Kconfig conflicts are resolved. You would encounter the same issue in your build sdkconfig values for example.idf confserver.py

I would recommend to open an issue in the [esp-sr component repository](https://github.com/espressif/esp-sr)

sun-xiangyu, 2025-04-25:
Hi @nukerl , 感谢你的反馈

考虑到后面这两个模型(wakenet9 and wakenet9s)选项可能会同时存在,我会修改这个menu的名字

MyCodeArea, 2025-04-28:
在提自己的issue时看到这里。我也遇到了类似问题。我遇到的情况是Load Multiple Wake Words的菜单就不会出现。将两个分别改为Load Multiple Wake Words1和Load Multiple Wake Words2后，能根据依赖项正常出现。

sun-xiangyu, 2025-04-28:
我已经修改为 Load Multiple Wake Words (WakeNet9s) 和 Load Multiple Wake Words (WakeNet9)