espressif/esp-sr #145: ESP32(非ESP32-S3)运行ESP-SR V2.0.5 Wakenet9 唤醒词检测，CPU占用率达到了39%

MyCodeArea, 2025-04-28:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

你好。
我们在ESP32（非ESP32-S3）运行ESP-SR V2.0.5。根据新的改变，ESP32支持Wakenet9，也不再支持Wakenet5。只做一个唤醒词的检测（目前测试的是hilexin），CPU占用率常态处于28%（按照双核总共运算能力统计）以上，最高峰峰值达到了39%。这导致其运行所在的cpu core长期idle线程CPU占用率接近0（还有其他的任务占用该core），且时常出现看门狗超时提醒。 而此前我们运行Wakenet5， 占用率常态地在8%左右，峰值也只有10%。由于我们的设备需要运行较多其他的任务，所以想了解WN9是否有优化这一部分的可能，还是因为我们的配置有些问题？
一些配置参数的log如下：

pcm_config.total_ch_num: 1udio Front End) Parameter **********/
pcm_config.mic_num: 1: [ ch0 ]
pcm_config.ref_num: 0: [  ]
pcm_config.sample_rate: 16000
afe_type: SR
afe_mode: HIGH PERF
afe_perferred_core: 0
afe_perferred_priority: 4
afe_ringbuf_size: 50
memory_alloc_mode: 3
afe_linear_gain: 1.0
debug_init: falsefixed_first_channel: false

/********** AEC (Acoustic Echo Cancellatio **********/
aec_init: false
aec mode: SR_LOW_COST
aec_filter_length: 4

/********** SE (Speech Enhancement, Microphone Array Processing) **********/
se_init: true, model: BSS

/********** NS (Noise Suppression) **********/
ns_init: false
ns model name: WEBRTC

/********** VAD (Voice Activity Detection) **********/
vad_init: true
vad_mode: 3
vad_model_name: NULL
vad_min_speech_ms: 128
vad_min_noise_ms: 1000
vad_delay_ms: 128
vad_mute_playback: false
vad_enable_channel_trigger: false

/********** WakeNet (Wake Word Engine) **********/
wakenet_init: true
wakenet_model_name: wn9_hilexin
wakenet_model_name_2: NULL
wakenet_mode: 0

/********** AGC (Automatic Gain Control) **********/
agc_init: false
agc_mode: WAKENET
agc_compression_gain_db: 9
agc_target_level_dbfs: 9

/**************************************************/
W (13213) AFE_CONFIG: For single microphone channel, SE is deactivated.
I (13263) Application: audio input rate:15996 samples/s
MC Quantized wakenet9: wakenet9_v1h24_嗨，乐鑫_3_0.608_0.615, tigger:v3, mode:0, p:0, (Apr 17 2025 15:49:41)
I (13323) AFE: AFE Version: (1MIC_V250121)
I (13323) AFE: Input PCM Config: total 1 channels(1 microphone, 0 playback), sample rate:16000
I (13333) AFE: AFE Pipeline: [input] -> |VAD(WebRTC)| -> |WakeNet(wn9_hilexin,)| -> [output]

sun-xiangyu, 2025-04-28:
@MyCodeArea , 是的wakenet9 在esp32上运行确实需要这么多CPU loading。 
我们开发了一个新的wakenet9s 会在不支持SIMD的芯片上有更合理的速度，现在还没有支持esp32， 会在下一个版本中支持。

sun-xiangyu, 2025-04-29:
预计3周以后