espressif/esp-sr #146: 可否不使用 "乐鑫语音唤醒方案客户定制" ，自己在本地训练唤醒模型？

MythHM, 2025-05-02:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

可否不使用 "乐鑫语音唤醒方案客户定制" ，自己在本地训练唤醒模型？

sun-xiangyu, 2025-05-12:
基于目前的开源方案，你可以做任意修改，包括使用自己的唤醒词模型

shuofeng1, 2025-05-14:
> 基于目前的开源方案，你可以做任意修改，包括使用自己的唤醒词模型
请问当前使用的唤醒词模型有提供开源自训练吗

shuofeng1, 2025-05-14:
> 基于目前的开源方案，你可以做任意修改，包括使用自己的唤醒词模型

@sun-xiangyu 请问现在使用的唤醒词模型有提供开源自训练吗

sun-xiangyu, 2025-05-14:
没有

MythHM, 2025-05-19:
> 基于目前的开源方案，你可以做任意修改，包括使用自己的唤醒词模型

那要如何在本地训练唤醒模型，我看文档中没有给出具体方法，只有官方的自定义。

sun-xiangyu, 2025-05-19:
> > 基于目前的开源方案，你可以做任意修改，包括使用自己的唤醒词模型
> 
> 那要如何在本地训练唤醒模型，我看文档中没有给出具体方法，只有官方的自定义。

我的意思是你可以用你自己的模型，或是网上其他开源的唤醒模型。我们的唤醒模型训练部分目前是没有开源的

98ytghhs, 2025-12-03:
我们正在研发一款基于esp32s3做一款AI心理陪伴，主要解决学生心理问题；名称是“蒲娃”，请您帮我们训练一个“你好 蒲娃”非常感谢 联系方式：2193558501@qq.com

sun-xiangyu, 2025-12-03:
@98ytghhs ， 如果申请唤醒词模型，请在https://github.com/espressif/esp-sr/issues/88 中留言，并注意规则

98ytghhs, 2025-12-03:
好的，看到了，就在里面申请，感谢

closed 2026-04-03 by feizi