espressif/esp-sr #148: Failed to build for esp32c5

jiruffe, 2025-05-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I tried [https://github.com/espressif/esp-sr/blob/master/test_apps/esp32c5/main/test_aec.cpp](esp32c5/main/test_aec.cpp) to use AEC on ESP32-C5 but failed to build.

### Actual behavior (suspected bug)

See logs below.

### Error logs or terminal output

```shell
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: D:/Documents/projects/example/managed_components/espressif__esp-sr/lib/esp32c5/libesp_audio_front_end.a(esp_afe_aec.c.obj): in function `afe_aec_process':
/home/sunxiangyu/workspace/esp_sr_lib_c5/components/esp_audio_front_end/esp_afe_aec.c:130:(.text.afe_aec_process+0x60): undefined reference to `aec_process'
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: D:/Documents/projects/example/managed_components/espressif__esp-sr/lib/esp32c5/libesp_audio_front_end.a(esp_afe_aec.c.obj): in function `afe_aec_destroy':
/home/sunxiangyu/workspace/esp_sr_lib_c5/components/esp_audio_front_end/esp_afe_aec.c:156:(.text.afe_aec_destroy+0x26): undefined reference to `aec_destroy'
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: D:/Documents/projects/example/managed_components/espressif__esp-sr/lib/esp32c5/libesp_audio_front_end.a(esp_afe_aec.c.obj): in function `afe_aec_create':
/home/sunxiangyu/workspace/esp_sr_lib_c5/components/esp_audio_front_end/esp_afe_aec.c:110:(.text.afe_aec_create+0x56): undefined reference to `aec_create'
C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: /home/sunxiangyu/workspace/esp_sr_lib_c5/components/esp_audio_front_end/esp_afe_aec.c:110:(.text.afe_aec_create+0x60): undefined reference to `aec_get_chunksize'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

### Steps to reproduce the behavior

1. `idf.py add-dependency "espressif/esp-sr^2.1.0"`
2. copy [https://github.com/espressif/esp-sr/blob/master/test_apps/esp32c5/main/test_aec.cpp](esp32c5/main/test_aec.cpp)
3. build
4. see errors

### Project release version

2.1.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

ZSH

### Additional context

ESP-IDF 5.4.0
ESP-SR 2.1.0

sun-xiangyu, 2025-05-06:
I've tested it on the Linux system and it works well. I will check it on the Windows 11.

jiruffe, 2025-05-11:
@sun-xiangyu Hello, please check #150. It works for me.

closed 2025-05-13 by wujiangang

Lucia-Rubio-E, 2025-05-22:
Hi @jiruffe,

Quick question — do you happen to know where I could buy an ESP32-C5 module or chip with ECO2 (v1.1 or v1.2 silicon)? I'm having trouble sourcing one.

Really appreciate your help!

sun-xiangyu, 2025-05-22:
Please try https://www.espressif.com/zh-hans/contact-us/get-samples

commit 095cb4cf (2025-05-13): bugfix: fixed #148
PR #150 (2025-05-13): bugfix: fixed #148 (AIS-1999)

## Description

## Related

## Tes