espressif/esp-sr #155: multinet->create(mn_name, duration)，duration有上限值吗，太大会一直ESP_MN_STATE_TIMEOUT状态

VSiR22, 2025-06-09:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

multinet->create(mn_name, duration); 当duration大于一定值时，如500000，会一直是ESP_MN_STATE_TIMEOUT状态

VSiR22, 2025-06-11:
设置：
int wakeup_time = 500000;
model_iface_data_t *model_data = multinet->create(mn_name, wakeup_time);
唤醒后打印结果：
WAKEWORD DETECTED
timeout, string:（唤醒后立刻打印该行）

-----------awaits to be waken up-----------

sun-xiangyu, 2025-06-11:
你用的是哪个模型? duration在代码中没有设置上限，如果出现你的问题，可能是bug

VSiR22, 2025-06-11:
> 你用的是哪个模型? duration在代码中没有设置上限，如果出现你的问题，可能是bug

**模型是general chinese recognition(mn7 cn)；**使用的是esp-skainet中的cn_speech_commands_recognition例程；
设置为一直唤醒的状态，但是每过duration时间会进入ESP_MN_STATE_TIMEOUT状态；如果刚好这时说出命令会识别不到；
所以想法是设置duration为一个较大的值，之后就发现当duration大于一个阈值的时候就会直接进入ESP_MN_STATE_TIMEOUT状态

sun-xiangyu, 2025-06-12:
因为在计算一个变量的时候溢出了 int 类型的最大值

closed 2025-06-17 by feizi