espressif/esp-sr #157: ESP32S3 当我使用唤醒词唤醒（WAKENET）时没有问题，但切到命令识别（MULTINET）时，不断输出W (49977) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting 并且输出错误的结果，TOP 1, command_id: 0, phrase_id: 0, string: da da kai tai deng prob: 0.326529

hulife, 2025-06-17:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

我的代码如下：
`#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "nvs_flash.h"
#include "my_spi.h"
#include "led.h"
#include "myiic.h"
#include "xl9555.h"
#include "spilcd.h"
#include "spi_sd.h"
#include "sdmmc_cmd.h"
#include "text.h"
#include "fonts.h"
#include "key.h"
#include "text.h"
#include "fonts.h"
#include "es8388.h"
#include "myi2s.h"
#include "exfuns.h"
#include "audioplay.h"
#include "recorder.h"
#include <stdio.h>

#include "esp_wn_iface.h"
#include "esp_wn_models.h"
#include "esp_afe_sr_models.h"
#include "esp_mn_iface.h"
#include "esp_mn_models.h"
#include "model_path.h"

#include "esp_afe_sr_iface.h"
#include "esp_mn_speech_commands.h"

#define TAG "app"
typedef struct {
    wakenet_state_t     wakenet_mode;
    esp_mn_state_t      state;
    int                 command_id;
} sr_result_t;

static QueueHandle_t            g_result_que    = NULL;
int detect_flag = 0;
static volatile int task_flag = 0;

static esp_afe_sr_iface_t *afe_handle = NULL;
static esp_afe_sr_data_t *afe_data = NULL;

//static model_iface_data_t       *model_data     = NULL;
//static const esp_mn_iface_t     *multinet       = NULL;
srmodel_list_t *models = NULL;
const char *cmd_phoneme[10] = {
    "da kai tai deng",
    "guan bi tai deng",
    "tai deng tiao liang",
    "tai deng tiao an",
    "da kai deng dai",
    "guan bi deng dai",
    "bo fang yin yue",
    "ting zhi bo fang",
    "da kai shi jian",
    "da kai ri li"
};
[... 169 lines cut ...]
    // models：声学前端和麦克风唤醒模型列表。
    // AFE_TYPE_VC :用于语音通话降噪 AFE_TYPE_SR：用于语音识别。
    // AFE_MODE_LOW_COST：AFE 模式，低功耗模式。
    afe_config_t *afe_config = afe_config_init("MR", models, AFE_TYPE_SR, AFE_MODE_HIGH_PERF);
    //afe_config->afe_ringbuf_size = 4096; // Increase from default value
    // print/modify wake word model.
    if (afe_config->wakenet_model_name)
    {
        printf("wakeword model in AFE config: %s\n", afe_config->wakenet_model_name);
    }
    if (afe_config->wakenet_model_name_2)
    {
        printf("wakeword model in AFE config: %s\n", afe_config->wakenet_model_name_2);
    }

    afe_handle = esp_afe_handle_from_config(afe_config);
    esp_afe_sr_data_t *afe_data = afe_handle->create_from_config(afe_config);
    afe_config_free(afe_config); // 释放 AFE 配置内存
    task_flag = 1; // 设置任务标志为1，启动任务
    // 创建检测唤醒词任务，运行在核心1
    xTaskCreatePinnedToCore(&detect_Task, "detect", 8 * 1024, (void *)afe_data, 8, NULL, 0);
    // 创建音频采集任务，运行在核心0
    xTaskCreatePinnedToCore(&feed_Task, "feed", 8 * 1024, (void *)afe_data, 5, NULL, 1);

}`

feizi, 2025-06-17:
> W (83948) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting
> W (83978) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting
> W (84008) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting
> W (84053) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting
> W (84097) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting
> W (84127) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting
> W (84157) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting`

这个log说明芯片现在的负载太大， fetch 被阻塞住了，导致丢数据。
建议： 
1. 可以选择 mn7 模型，这个模型更快一点
2. 检查一下kconfig, 对齐 [examples/cn_speech_commands_recognition](https://github.com/espressif/esp-skainet/tree/master/examples/cn_speech_commands_recognition), 可以设置更大的cache size 和 cache line
3. 将 multinet 移到单独的核上

hulife, 2025-06-18:
> > W (83948) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting
> > W (83978) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting
> > W (84008) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting
> > W (84053) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting
> > W (84097) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting
> > W (84127) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting
> > W (84157) AFE: Ringbuffer of AFE(FEED) is full, Please use fetch() to read data to avoid data loss or overwriting`
> 
> 这个log说明芯片现在的负载太大， fetch 被阻塞住了，导致丢数据。 建议：
> 
> 1. 可以选择 mn7 模型，这个模型更快一点
> 2. 当唤醒后，使用 afe_handle->disable_wakenet(afe_data); 暂时关闭 wakenet, 退出时 afe_handle->enable_wakenet(afe_data); 再次开始
> 3. 检查一下kconfig, 对齐 [examples/cn_speech_commands_recognition](https://github.com/espressif/esp-skainet/tree/master/examples/cn_speech_commands_recognition), 可以设置更大的cache size 和 cache line
> 4. 将 multinet 移到单独的核上

**我现在选择了1.mn7模型 2.暂时关闭 wakenet 3.增大了cache size 和 cache line 4.单独的核，现在现象是仍然出现Ringbuffer of AFE(FEED) is full,但是不会一直打印了，可是唤醒词仍然是生效的，但命令词仍不识别：

日志如下：**
27) boot: ESP-IDF v5.4.1-dirty 2nd stage bootloader
I (27) boot: compile time Jun 18 2025 09:12:42
I (27) boot: Multicore bootloader
I (28) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (34) qio_mode: Enabling default flash chip QIO
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (42) boot.esp32s3: SPI Mode       : QIO
I (46) boot.esp32s3: SPI Flash Size : 16MB
I (50) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 factory          factory app      00 00 00010000 00200000
I (70) boot:  1 model            Unknown data     01 82 00210000 0050c000
I (76) boot: End of partition table
I (79) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=233a4h (144292) map
I (108) esp_image: segment 1: paddr=000333cc vaddr=3fc9be00 size=060a8h ( 24744) load
I (113) esp_image: segment 2: paddr=0003947c vaddr=40378000 size=06b9ch ( 27548) load
I (118) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=90ce4h (593124) map
I (208) esp_image: segment 4: paddr=000d0d0c vaddr=4037eb9c size=0d18ch ( 53644) load
I (218) esp_image: segment 5: paddr=000ddea0 vaddr=600fe100 size=0001ch (    28) load
I (227) boot: Loaded app from partition at offset 0x10000
I (227) boot: Disabling RNG early entropy source...
I (237) octal_psram: vendor id    : 0x0d (AP)
I (237) octal_psram: dev id       : 0x02 (generation 3)
I (237) octal_psram: density      : 0x03 (64 Mbit)
I (239) octal_psram: good-die     : 0x01 (Pass)
I (243) octal_psram: Latency      : 0x01 (Fixed)
I (248) octal_psram: VCC          : 0x01 (3V)
I (252) octal_psram: SRF          : 0x01 (Fast Refresh)
I (257) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (262) octal_psram: BurstLen     : 0x01 (32 Byte)
I (266) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (271) octal_psram: DriveStrength: 0x00 (1/1)
I (276) MSPI Timing: PSRAM timing tuning index: 5
I (280) esp_psram: Found 8MB PSRAM device
I (284) esp_psram: Speed: 80MHz
I (287) cpu_start: Multicore app
I (591) esp_psram: SPI SRAM memory test OK
I (600) cpu_start: Pro cpu start user code
I (600) cpu_start: cpu freq: 240000000 Hz
I (600) app_init: Application information:
I (600) app_init: Project name:     31_recoding
I (604) app_init: App version:      1
I (607) app_init: Compile time:     Jun 18 2025 09:27:38
I (612) app_init: ELF file SHA256:  8ad53b860...
I (617) app_init: ESP-IDF:          v5.4.1-dirty
I (621) efuse_init: Min chip rev:     v0.0
I (625) efuse_init: Max chip rev:     v0.99 
I (629) efuse_init: Chip rev:         v0.2
I (633) heap_init: Initializing. RAM available for dynamic allocation:
I (639) heap_init: At 3FCA2DC0 len 00046950 (282 KiB): RAM
[... 90 lines cut ...]
-----------awaits to be waken up-----------
WAKEWORD DETECTED........
model index:1, word index:1
timeout, string:

-----------awaits to be waken up-----------
WAKEWORD DETECTED........
model index:1, word index:1
timeout, string:qi

-----------awaits to be waken up-----------
WAKEWORD DETECTED........
model index:1, word index:1
timeout, string:

-----------awaits to be waken up-----------
WAKEWORD DETECTED........
model index:1, word index:1
timeout, string:

我的配置如下：

[image]

[image]

hulife, 2025-06-18:
我将任务运行的内核由
 xTaskCreatePinnedToCore(&detect_Task, "detect", 8 * 1024, (void *)afe_data, 8, NULL, 1);
 xTaskCreatePinnedToCore(&feed_Task, "feed", 8 * 1024, (void *)afe_data, 5, NULL, 0);
改为
xTaskCreatePinnedToCore(&detect_Task, "detect", 8 * 1024, (void *)afe_data, 8, NULL, 0);
xTaskCreatePinnedToCore(&feed_Task, "feed", 8 * 1024, (void *)afe_data, 5, NULL, 1);
或者
xTaskCreatePinnedToCore(&detect_Task, "detect", 8 * 1024, (void *)afe_data, 8, NULL, 1);
xTaskCreatePinnedToCore(&feed_Task, "feed", 8 * 1024, (void *)afe_data, 5, NULL, 1);
或者
xTaskCreatePinnedToCore(&detect_Task, "detect", 8 * 1024, (void *)afe_data, 8, NULL, 0);
xTaskCreatePinnedToCore(&feed_Task, "feed", 8 * 1024, (void *)afe_data, 5, NULL, 0);
前几秒出现的溢出现象Ringbuffer of AFE(FEED) is full,消失了，我的示例中只运行这两个任务，为什么会有这种情况呢？（唤醒生效，命令词不生效现象仍有）

sun-xiangyu, 2025-06-19:
可以使用[mutlinet->set_det_threshold(model_data, threshold)](https://github.com/espressif/esp-sr/blob/75287ca5a7d5b4034e055f2f2682884577113fd2/include/esp32s3/esp_mn_iface.h#L118) 调节阈值

hulife, 2025-06-19:
> 可以使用[mutlinet->set_det_threshold(model_data, threshold)](https://github.com/espressif/esp-sr/blob/75287ca5a7d5b4034e055f2f2682884577113fd2/include/esp32s3/esp_mn_iface.h#L118) 调节阈值

降低命令词阈值到0.05之后，部分命令可以识别了，但是还是不是特别灵敏，有些指令反复说出才能识别，唤醒词阈值我设置的0.5，每次识别挺快的

-----------listening-----------
TOP 1, command_id: 5, phrase_id: 5, string: qian jin prob: 0.056983
-----------listening-----------
TOP 1, command_id: 6, phrase_id: 6, string: da kai ri li prob: 0.050314
-----------listening-----------
TOP 1, command_id: 5, phrase_id: 5, string: qian jin prob: 0.082873
-----------listening-----------
TOP 1, command_id: 4, phrase_id: 4, string: hou tui prob: 0.132107

为什么命令词和唤醒词识别时差距这么大呢？相同环境下有什么因素会影响命令词识别吗？

sun-xiangyu, 2025-06-19:
两个模型不一样，命令词模型识别的是单个音素(拼音单元)，多个因素的概率累加在一起就会比较低

closed 2025-12-04 by feizi