espressif/esp-sr #165: ESP32-S3 Crash During Wake Word Model Initialization (LoadProhibited Exception, wakenet9)

abdulbaseer-1, 2025-07-28:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

i expect wakenet9 to initialize correctly .

### Actual behavior (suspected bug)

**issue:(!pls help I have been trying to solve this since 2 week)

Using ESP32-S3 with ESP-IDF v5.5 and esp-sr library v2.0.

Speech recognition models stored in a flash partition labeled "model" at offset 0x410000, size 8MB.

Partition is successfully found and memory-mapped.

Model data pointer (wn_data->data[0]) appears valid and outputs correct hex data.

Crash occurs immediately during wakenet->create() call inside esp-sr library (wakenet9s_quantized_mc.c line 202).

Exception: LoadProhibited (memory access violation).

Backtrace points to model_destroy() function during model initialization.

Application reboots after crash. 
 
 
 
 
 
 
 
 
Summary of ESP32-S3 Wake Word Detection Crash Issue 

Environment & Setup
Device: ESP32-S3

ESP-IDF Version: v5.5-dirty (built Jul 28, 2025)

esp-sr (Speech Recognition) library version: v2.0.0 (latest)

Flash Size: 16 MB external SPI flash

Partition Table:

nvs: 0x9000, size 0x4000

otadata: 0xd000, size 0x2000

app0 (factory app): 0x10000, size 4MB

model (custom data): 0x410000, size 8MB

Description
The project uses a multinet + wakenet speech model approach for wake word detection and command recognition.

Models are stored in the "model" partition, loaded via esp_srmodel_init("model") which uses flash memory mmap to map the partition.

The partition is successfully mmap’d, and initial logs indicate models are loaded correctly:

yaml
[... 348 lines cut ...]

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

FreeRTOS

### Shell

CMD

### Additional context

System Architecture :xtensa cpu=>esp32s3 
  
 
i am a student
please-help!!! 
i-will-fail-my-project.

sun-xiangyu, 2025-07-28:
> wn_handle = wakenet->create((const void *)wn_data->data[0], WAKE_MODE);

The first parameter of create function is model_name. Please refer to the following code or [esp-skainet example](https://github.com/espressif/esp-skainet/blob/master/examples/wake_word_detection/wakenet/main/main.c#L21C1-L24C79)

```
    srmodel_list_t *models = esp_srmodel_init("model");
    char *model_name = esp_srmodel_filter(models, ESP_WN_PREFIX, "hilexin");
    esp_wn_iface_t *wakenet = (esp_wn_iface_t*)esp_wn_handle_from_name(model_name);
    model_iface_data_t *model_data = wakenet->create(model_name, DET_MODE_95);
```

sun-xiangyu, 2025-07-29:
Do not need `srmodel_data_t *wn_data = sr_models->model_data[wn_index];`, Just input model name

sun-xiangyu, 2025-07-29:
The code doesn't seem to have any issues. Which version of esp-sr are you using? The wn9s model is only supported from v2.1.0 onwards.

abdulbaseer-1, 2025-07-29:
Respected Sir, 
I am using esp-idf v5.5 with esp-sr v2.1.4. and still getting the same issue as above. 
below is my updated github repo with complete build available (complete!).I have changed the model to wn9_hilexin, but still same issue. 
**I thank you greatly for your help.**  
 
github_repo : **https://github.com/abdulbaseer-1/ESP32_chatbot_hijason.git** 
 
Respectfully, 
Abdul Baseer.

sun-xiangyu, 2025-07-30:
Hi, I don't have time to learn your project unless you can reproduce your issue with the shortest code.

abdulbaseer-1, 2025-08-02:
**Respected Sir, Please help The only portion of code related to the error has been provided in the above comment**  
**Thank you greatly for your help**
**Respectfully,** 
**Abdul Baseer**