espressif/esp-sr #166: ESP32-S3 Crash During Wake Word Model Initialization (LoadProhibited Exception, wakenet9) (IDFGH-16103)

abdulbaseer-1, 2025-07-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v 5.5

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3

### Power Supply used.

USB

### What is the expected behavior?

i expect the model to create successfully.

### What is the actual behavior?

the model is loaded but it fails in **wn_handle = wakenet->create((const void *)wn_data->data[0], WAKE_MODE);** 
everything seems okay but it still fails.

### Steps to reproduce.

just build and flash the project for your esp32s3. 
 
make sure to install esp-sr. 
 
repo:https://github.com/abdulbaseer-1/ESP32_chatbot_hijason.git 
 
Note:now i am using wn9s model just to make sure it was not a PSRAM issue.

**#include "esp_wn_models.h" 
//#include "esp_mn_models.h"
#include "wakeword.h"
#include "mic_i2s.h"
#include "esp_log.h"
#include "model_path.h" // Replace esp_srmodel.h with model_path.h
#include "esp_wn_iface.h"
//#include "esp_mn_iface.h"
//#include "esp_mn_speech_commands.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

static const char *TAG = "WakeMultinet";

#define SAMPLE_RATE     16000
#define BUFFER_SIZE     512
#define WAKE_MODE       DET_MODE_95
[... 179 lines cut ...]
--- 0x400556d5: strlen in ROM
--- 0x400556e5: strlen in ROM

Backtrace: 0xfffffffd:0x3fcaa760 0x4201161d:0x3fcaa780 0x42011634:0x3fcaa940 0x4200c0d4:0x3fcaa960 0x4200bc7c:0x3fcaa990 0x420c0ae0:0x3fcaa9d0 0x4037ddc5:0x3fcaaa00
--- 0x4201161d: model_create at /home/sunxiangyu/workspace/esp_sr_lib/components/wakenet/wakenet9s_quantized_mc.c:512
--- 0x42011634: model_init at /home/sunxiangyu/workspace/esp_sr_lib/components/wakenet/wakenet9s_quantized_mc.c:904
--- 0x4200c0d4: wakeword_init_multinet at F:/personal/internship/esp32-voice-assistant/firmware_SR/esp32_chatbot/components/custom_audio/wakeword.c:65
--- 0x4200bc7c: app_main at F:/personal/internship/esp32-voice-assistant/firmware_SR/esp32_chatbot/main/main.c:94
--- 0x420c0ae0: main_task at F:/personal/internship/esp32-voice-assistant/firmware_SR/esp32_chatbot/components/freertos/app_startup.c:208
--- 0x4037ddc5: vPortTaskWrapper at F:/personal/internship/esp32-voice-assistant/firmware_SR/esp32_chatbot/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: 33560f4d8
```

### Diagnostic report archive.

[idf-diag-059cd62d-0d64-4359-98df-a3d194ec04ff.zip](https://github.com/user-attachments/files/21482490/idf-diag-059cd62d-0d64-4359-98df-a3d194ec04ff.zip)

### More Information.

Everything seems normal but the issue still persists. 
I have been trying to solve this since 3 days.
i am a student
please-help!!!
i-will-fail-my-project.

vishaltech21, 2025-08-02:
I am also trying to do wake word detection using esp32 and i2s mic,

my arduino code: "#include <stdio.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_wn_iface.h"
#include "esp_wn_models.h"
#include "model_path.h"
#include "string.h"
#include "driver/i2s.h"

#define SAMPLE_RATE     16000
#define BITS_PER_SAMPLE I2S_BITS_PER_SAMPLE_16BIT
#define I2S_PORT        I2S_NUM_1
#define CHANNELS        1
#define RECORD_DURATION_SECONDS 3
#define BUFFER_SIZE     512
#define DMA_BUF_COUNT   8

typedef struct {
    const esp_wn_iface_t *wakenet;
    const char *model_name;
} wakeword_config_t;

void wakeword_task(void *arg)
{
    wakeword_config_t *config = (wakeword_config_t *)arg;

    if (!config || !config->wakenet || !config->model_name) {
        Serial0.println("ERROR: Invalid wakeword task config");
        vTaskDelete(NULL);
        return;
    }

    const esp_wn_iface_t *wakenet = config->wakenet;
    const char *model_name = config->model_name;

    model_iface_data_t *model_data = wakenet->create(model_name, DET_MODE_95);
    if (!model_data) {
        Serial0.println("ERROR: Failed to create WakeNet model instance");
        vTaskDelete(NULL);
        return;
    }

    int chunk_size = wakenet->get_samp_chunksize(model_data);
    int16_t *buffer = (int16_t *)malloc(chunk_size * sizeof(int16_t));
    if (!buffer) {
        Serial0.println("ERROR: Failed to allocate buffer");
        wakenet->destroy(model_data);
        vTaskDelete(NULL);
        return;
    }

    Serial0.println("WakeNet task running, waiting for wake word...");

    while (1) {
        size_t bytes_read = 0;
        esp_err_t res = i2s_read(I2S_PORT, buffer, chunk_size * sizeof(int16_t), &bytes_read, portMAX_DELAY);
        if (res == ESP_OK && bytes_read > 0) {
            
            // 👂 Wake word detection
            wakenet_state_t state = wakenet->detect(model_data, buffer);
            if (state == WAKENET_DETECTED || state == 1) {
                Serial0.println("Wake word detected!");
            }
        } else {
            Serial0.println("WARNING: i2s_read failed or no data");
        }
    }

    free(buffer);
    wakenet->destroy(model_data);
    vTaskDelete(NULL);
}

void setupI2S() {
  i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX),
    .sample_rate = SAMPLE_RATE,
    .bits_per_sample = BITS_PER_SAMPLE,
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
    .communication_format = I2S_COMM_FORMAT_I2S,
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = DMA_BUF_COUNT,
    .dma_buf_len = BUFFER_SIZE,
    .use_apll = false,
    .tx_desc_auto_clear = false
  };

  i2s_pin_config_t pin_config = {
    .bck_io_num = 42,
    .ws_io_num = 1,
    .data_out_num = I2S_PIN_NO_CHANGE,
    .data_in_num = 2
  };

  i2s_driver_install(I2S_PORT, &i2s_config, 0, NULL);
  i2s_set_pin(I2S_PORT, &pin_config);
  i2s_zero_dma_buffer(I2S_PORT);
}

void setup() {
    Serial0.begin(115200);
    delay(1000); // Allow USB CDC startup

    Serial0.println("Setup starting...");

    setupI2S();

    srmodel_list_t *models = esp_srmodel_init("model");
    if (!models) {
        Serial0.println("ERROR: Failed to initialize speech models");
        return;
    }
    char *model_name = esp_srmodel_filter(models, ESP_WN_PREFIX, "jarvis");
    if (!model_name) {
        Serial0.println("ERROR: Wake word model 'jarvis' not found");
        return;
    }

    Serial0.print("Model selected: ");
    Serial0.println(model_name);

    const esp_wn_iface_t *wakenet = esp_wn_handle_from_name(model_name);
    if (!wakenet) {
        Serial0.println("ERROR: Failed to get WakeNet handle");
        return;
    }

    static wakeword_config_t config;
    config.wakenet = wakenet;
    config.model_name = model_name;

    xTaskCreate(wakeword_task, "wakeword_task", 8192, (void *)&config, 5, NULL);

}

void loop() {
    delay(100);
}
"

output:
E (Setup starting...
Model selected: wn9_jarvis_tts
MC Quantized wakenet9: wakenet9l_tts1h8_Jarvis_3_0.627_0.632
, tigger:v3, mode:1, p:0, (Nov  5 2024 16:02:49)
WakeNet task running, waiting for wake word...

But the problem is it not detecting wake word correctly even if i tell the wake word "jarvis".

vishaltech21, 2025-08-02:
try this code, Just now I fixed and Working well!

#include <stdio.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_wn_iface.h"
#include "esp_wn_models.h"
#include "model_path.h"
#include "string.h"
#include "driver/i2s.h"

#define SAMPLE_RATE     16000
#define BITS_PER_SAMPLE I2S_BITS_PER_SAMPLE_16BIT
#define I2S_PORT        I2S_NUM_1
#define CHANNELS        1
#define RECORD_DURATION_SECONDS 3
#define BUFFER_SIZE     512
#define DMA_BUF_COUNT   8

typedef struct {
    const esp_wn_iface_t *wakenet;
    const char *model_name;
} wakeword_config_t;

void wakeword_task(void *arg)
{
    wakeword_config_t *config = (wakeword_config_t *)arg;

    if (!config || !config->wakenet || !config->model_name) {
        Serial0.println("ERROR: Invalid wakeword task config");
        vTaskDelete(NULL);
        return;
    }

    const esp_wn_iface_t *wakenet = config->wakenet;
    const char *model_name = config->model_name;

    model_iface_data_t *model_data = wakenet->create(model_name, DET_MODE_95);
    if (!model_data) {
        Serial0.println("ERROR: Failed to create WakeNet model instance");
        vTaskDelete(NULL);
        return;
    }

    int chunk_size = wakenet->get_samp_chunksize(model_data);
    int16_t *buffer = (int16_t *)malloc(chunk_size * sizeof(int16_t));
    if (!buffer) {
        Serial0.println("ERROR: Failed to allocate buffer");
        wakenet->destroy(model_data);
        vTaskDelete(NULL);
        return;
    }

    Serial0.println("WakeNet task running, waiting for wake word...");

    while (1) {
        size_t bytes_read = 0;
        esp_err_t res = i2s_read(I2S_PORT, buffer, chunk_size * sizeof(int16_t), &bytes_read, portMAX_DELAY);
        if (res == ESP_OK && bytes_read > 0) {

            // 👂 Wake word detection
            wakenet_state_t state = wakenet->detect(model_data, buffer);
            if (state == WAKENET_DETECTED || state == 1) {
                Serial0.println("Wake word detected!");
            }
        } else {
            Serial0.println("WARNING: i2s_read failed or no data");
        }
    }

    free(buffer);
    wakenet->destroy(model_data);
    vTaskDelete(NULL);
}

void setupWakeWord() {
    srmodel_list_t *models = esp_srmodel_init("model");
    if (!models) {
        Serial0.println("ERROR: Failed to initialize speech models");
        return;
    }
    char *model_name = esp_srmodel_filter(models, ESP_WN_PREFIX, "jarvis");
    if (!model_name) {
        Serial0.println("ERROR: Wake word model 'jarvis' not found");
        return;
    }

    Serial0.print("Model selected: ");
    Serial0.println(model_name);

    const esp_wn_iface_t *wakenet = esp_wn_handle_from_name(model_name);
    if (!wakenet) {
        Serial0.println("ERROR: Failed to get WakeNet handle");
        return;
    }

    static wakeword_config_t config;
    config.wakenet = wakenet;
    config.model_name = model_name;

    xTaskCreate(wakeword_task, "wakeword_task", 8192, (void *)&config, 5, NULL);
}

void setupI2S() {
  i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX),
    .sample_rate = SAMPLE_RATE,
    .bits_per_sample = BITS_PER_SAMPLE,
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
    .communication_format = I2S_COMM_FORMAT_I2S,
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = DMA_BUF_COUNT,
    .dma_buf_len = BUFFER_SIZE,
    .use_apll = false,
    .tx_desc_auto_clear = false
  };

  i2s_pin_config_t pin_config = {
    .bck_io_num = 42,
    .ws_io_num = 1,
    .data_out_num = I2S_PIN_NO_CHANGE,
    .data_in_num = 2
  };

  i2s_driver_install(I2S_PORT, &i2s_config, 0, NULL);
  i2s_set_pin(I2S_PORT, &pin_config);
  i2s_zero_dma_buffer(I2S_PORT);
}

void setup() {
    Serial0.begin(115200);
    delay(1000); // Allow USB CDC startup

    Serial0.println("Setup starting...");

    setupI2S();

    setupWakeWord();

}

void loop() {
    delay(100);
}