espressif/esp-sr #168: vad cache 里的音频数据，有时候全是静音

zeis333, 2025-08-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

often

### Expected behavior

VAD 触发后，当vad_cache_size大于0时，从vad_cache 里读取的音频应该是正常的音频数据。

### Actual behavior (suspected bug)

VAD 触发后，当vad_cache_size大于0时，从vad_cache 里读取的音频有时候是静音数据，导致上传云端后，云端识别的音频不完整，漏掉了说的第一个字。
在2.1.3及2.1.4版本测试都存在同样的问题。

出现问题的音频：
[image]

正常情况下的音频：
[image]

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

1. 唤醒词唤醒；
2. 正常说话触发VAD；
3. 获取云端识别结果；
4. 反复2.3步骤，有较高的概率会出现问题

### Project release version

sr-2.1.3,sr-2.1.4

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

win10

### Shell

ZSH

zeis333, 2025-08-07:
补充一下VAD 设置：

sun-xiangyu, 2025-08-07:
感谢反馈，我看看

aboyuanna, 2025-08-07:
> 补充一下VAD 设置：
> 
> ```
> /********** General AFE (Audio Front End) Parameter **********/
> pcm_config.total_ch_num: 4
> pcm_config.mic_num: 2: [ ch1, ch3 ]
> pcm_config.ref_num: 1: [ ch0 ]
> pcm_config.sample_rate: 16000
> afe_type: SR
> afe_mode: HIGH PERF
> afe_perferred_core: 0
> afe_perferred_priority: 5
> afe_ringbuf_size: 50
> memory_alloc_mode: 3
> afe_linear_gain: 1.0
> debug_init: false
> fixed_first_channel: false
> 
> /********** AEC (Acoustic Echo Cancellation) **********/
> aec_init: true
> aec mode: SR_HIGH_PERF
> aec_filter_length: 4
> 
> /********** SE (Speech Enhancement, Microphone Array Processing) **********/
> se_init: false, model: BSS
> 
> /********** NS (Noise Suppression) **********/
> ns_init: false
> ns model name: WEBRTC
> 
> /********** VAD (Voice Activity Detection) **********/
> vad_init: true
> vad_mode: 4
> vad_model_name: NULL
> vad_min_speech_ms: 256
> vad_min_noise_ms: 1200
> vad_delay_ms: 512
> vad_mute_playback: false
> vad_enable_channel_trigger: false
> ```

大佬，请问调用哪个函数可以打印这些参数出来？

sun-xiangyu, 2025-08-12:
@zeis333
请更新esp-sr, 再试一次，我修改了vad cache的复制逻辑。 由于我在测试的过程中没有复现过你这个会出现静音的bug，所以请你试一下是否解决了你的问题

zeis333, 2025-08-13:
@sun-xiangyu 我用master测试，同样的参数，没测到出现静音的问题了，这个修改会发布一个新的版本到组件库吗？

zeis333, 2025-08-13:
@sun-xiangyu 还有个情况请教一下，我当前使用的sr版本是2.1.3，当我更新sr版本到2.1.4或者当前的master后，我的应用代码都不改动的前提下，我的内部RAM可用空间减少20KB以上，这是正常的吗？

sun-xiangyu, 2025-08-13:
> [@sun-xiangyu](https://github.com/sun-xiangyu) 还有个情况请教一下，我当前使用的sr版本是2.1.3，当我更新sr版本到2.1.4或者当前的master后，我的应用代码都不改动的前提下，我的内部RAM可用空间减少20KB以上，这是正常的吗？

不正常，替换了FFT 的实现，可能会影响部分internal RAM， 但是不应该有 20KB
你主要使用了那些模块

zeis333, 2025-08-13:
> > [@sun-xiangyu](https://github.com/sun-xiangyu) 还有个情况请教一下，我当前使用的sr版本是2.1.3，当我更新sr版本到2.1.4或者当前的master后，我的应用代码都不改动的前提下，我的内部RAM可用空间减少20KB以上，这是正常的吗？
> 
> 不正常，替换了FFT 的实现，可能会影响部分internal RAM， 但是不应该有 20KB 你主要使用了那些模块

主要使用了AEC、VAD、WakeNet 、multinet，以下是功能参数，业务相关的信息用xxxxxx代替
```
/********** General AFE (Audio Front End) Parameter **********/
pcm_config.total_ch_num: 4
pcm_config.mic_num: 2: [ ch1, ch3 ]
pcm_config.ref_num: 1: [ ch0 ]
pcm_config.sample_rate: 16000
afe_type: SR
afe_mode: HIGH PERF
afe_perferred_core: 0
afe_perferred_priority: 5
afe_ringbuf_size: 50
memory_alloc_mode: 3
afe_linear_gain: 1.0
debug_init: false
fixed_first_channel: false

/********** AEC (Acoustic Echo Cancellation) **********/
aec_init: true
aec mode: SR_HIGH_PERF
aec_filter_length: 4

/********** SE (Speech Enhancement, Microphone Array Processing) **********/
se_init: false, model: BSS

/********** NS (Noise Suppression) **********/
ns_init: false
ns model name: WEBRTC

/********** VAD (Voice Activity Detection) **********/
vad_init: true
vad_mode: 4
vad_model_name: NULL
vad_min_speech_ms: 200
vad_min_noise_ms: 1200
vad_delay_ms: 512
vad_mute_playback: false
vad_enable_channel_trigger: false

/********** WakeNet (Wake Word Engine) **********/
wakenet_init: true
wakenet_model_name: wn9_xxxxxxxxx
wakenet_model_name_2: NULL
wakenet_mode: 0

/********** AGC (Automatic Gain Control) **********/
agc_init: false
agc_mode: WAKENET
agc_compression_gain_db: 9
agc_target_level_dbfs: 9

/**************************************************/
MC Quantized wakenet9: wakenet9_tts2h12v2_xxxxxxxxxxxxxxx, tigger:v4, mode:2, p:0, (Aug 11 2025 15:20:50)
I (2393) AFE: AFE Version: (2MIC_V250113)
I (2393) AFE: Input PCM Config: total 4 channels(2 microphone, 1 playback), sample rate:16000
I (2394) AFE: AFE Pipeline: [input] -> |AEC(SR_HIGH_PERF)| -> |VAD(WebRTC)| -> |WakeNet(wn9_xxxxxxxxx,)| -> [output]

multinet:mn7_en
Build fst from commands.
Quantized MultiNet7:rnnt_ctc_2.0, name:mn7_en, (Aug 11 2025 15:20:49)
Quantized MultiNet7 search method: 2, time out:6.0 s
Build fst from commands.
1 active speech commands:
Command 1: xxxxxxxxxxxxxx

```

zeis333, 2025-08-13:
@sun-xiangyu 从我测试的情况来看，离线命令功能影响比较大，占用的RAM空间变大很多

zeis333, 2025-08-14:
@sun-xiangyu IRAM占用情况，我重新开了一个问题。请问修改vad cache的版本，预计什么时候会发布到组件库呢？

sun-xiangyu, 2025-08-14:
已经发布了 2.1.5