espressif/esp-sr #173: crash when init at least two models

truth89757, 2025-09-05:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

在我的工程里我同时选择了multinet 和  vadnet ，在init的时候就会crash。

### Actual behavior (suspected bug)

必现

### Error logs or terminal output

```shell
--- 0x420d7f35: get_model_info at /Users/lulinjun/work_space/nertc-esp32/demo/esp32/managed_components/espressif__esp-sr/src/model_path.c:45
--- 0x420d8178: srmodel_load at /Users/lulinjun/work_space/nertc-esp32/demo/esp32/managed_components/espressif__esp-sr/src/model_path.c:300
--- 0x420d8269: srmodel_mmap_init at /Users/lulinjun/work_space/nertc-esp32/demo/esp32/managed_components/espressif__esp-sr/src/model_path.c:345
```

### Steps to reproduce the behavior

必现

### Project release version

lastest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

MacOS

### Operating system version

idf v5.4.2

### Shell

ZSH

truth89757, 2025-09-05:
model_data->files = (char **)malloc(sizeof(char *) * file_num);
        model_data->data = (char **)malloc(sizeof(void *) * file_num);
        model_data->sizes = (int *)malloc(sizeof(int) * file_num);
        printf("i =%d, model name: %s, file_num = %d\n", i, data, file_num); // 这里增加打印
        for (int j = 0; j < file_num; j++) {
--------------------------------------------------------------------------------------
实际输出为：
i =0, model name: vocab, file_num = 4
i =1, model name: wn9_data, file_num = 3
i =2, model name: commands_en.txt, file_num = 2
i =3, model name: vadn1_index, file_num = 3

truth89757, 2025-09-05:
build 的产物目录srmodels 里面包含 fst wn9_nihaoxiaozhi_tts mn7_cn vadnet1_medium srmodels.bin这五个文件

sun-xiangyu, 2025-09-22:
因为添加了 mn7_cn 比较大，注意partitions.csv中model 的分区应该给大一点，至少要比 srmodels.bin 要大

hushi55, 2025-12-02:
确实存在这个问题，使用 model/movemodel.py 
这个 python 打包后的 srmodels.bin 文件，必然奔溃
并且每次奔溃的位置的还不一样

truth89757, 2025-12-03:
确实是因为model的分区设置不够导致的崩溃，谢谢指出。@sun-xiangyu