espressif/esp-sr #174: 您好，是否可以出一个ES8311单麦克风的例程，我尝试修改双麦克风为单麦克风失败了。

caidamao, 2025-09-09:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

期望可以出一个使用ES8311+单麦克风的例程

### Use cases

我现在期望使用的是ESP S3+ES8311 挂载一个麦克风跟喇叭，由于成本问题，并不希望使用ES7210。
我尝试使用的环境ESP IDF V5.5.1 + ESP SR v2.0.0.
我现在根据查到的资料进行了一些尝试，但是不知道为什么一直没办法成功，并且找不到可以对照的例子，现在被困住了。我买了一块BOX3 但是这个是使用ES7210+双麦克风的，没有办法对照我问题出在哪里。
我现在的硬件使用ES8311里的i2s_echo是可以正常录音跟播放的。

caidamao, 2025-09-09:
[i2s_es8311_example.zip](https://github.com/user-attachments/files/22224607/i2s_es8311_example.zip)

sun-xiangyu, 2025-09-09:
@caidamao , 我对驱动也不是很熟悉，你可以在 [esp-adf](https://github.com/espressif/esp-adf) 或是[esp-bsp](https://github.com/espressif/esp-bsp) 中提问，他们可以提供关于驱动更准备的回答

caidamao, 2025-09-10:
> [@caidamao](https://github.com/caidamao) , 我对驱动也不是很熟悉，你可以在 [esp-adf](https://github.com/espressif/esp-adf) 或是[esp-bsp](https://github.com/espressif/esp-bsp) 中提问，他们可以提供关于驱动更准备的回答

我自己解决了这个问题，谢谢支持